v4rl-ucy/covins #4: An error occurred in the middle of the program

liming-jing, 2021-12-05:
Loading ORB Vocabulary. This could take a while...
Vocabulary loaded!

+++++ KeyFrame Database Initialized +++++
selectserver: new connection from 192.168.31.31 on socket 11
Pass new ID 0 to client
Map 0 : 50 KFs | 1495 LMs
Map 0 : 100 KFs | 2287 LMs
Map 0 : 150 KFs | 2515 LMs
double free or corruption (out)
Aborted (core dumped)

When running the data set, the program was directly terminated, causing the front end to be unable to link to the server, and the map creation failed. How should I solve it?
In addition, when I used ros to play the program, it also appeared that the data set was not played, and the server terminated the program. What should I do?

ThomasZiegler, 2021-12-13:
Hi
Can you also provide the logs from the agent? It looks like you only provided the server logs.

ThomasZiegler, 2021-12-13:
When you use the ros wrapper you need to play the rosbag containing the data set in a separate terminal window.

patriksc, 2021-12-18:
@liming-jing is this issue still relevant, or is the problem resolved in the meantime?

patriksc, 2022-01-08:
closed - inactive

closed 2022-01-08 by patriksc