v4rl-ucy/covins #5: problem about pointcloud

flyover-26, 2021-12-14:
hi!I noticed that this project is tested on Eoroc dataset.
And I didn't see the pointcloud result.So I want to konw that whether this project provides pointcloud result like ccm-slam?
And the second question is that if this project has a function of re-location?
Thank you!

patriksc, 2021-12-17:
Hi! If you run COVINS according to the manual and visualize the SLAM estimate using RVIZ with our provided instructions and config file, you should see the pointcloud, similar to the example in our publication video. Also, in #2, you can see another example.

Regarding your second question: COVINS does not provide any specific relocalization functionality after a tracking loss of the agent.

patriksc, 2022-07-31:
Problem seems resolved for the moment - closing this issue.

closed 2022-07-31 by patriksc