v4rl-ucy/covins #9: catkin build ORB_SLAM3 error

zhouleiqiang, 2021-12-29:
Hello, my system is ubuntu20.04 noetic. When I install to catkin build ORBSLAM3 according to your steps, this error is reported. Do you have a solution? Looking forward to your reply！！！！

Errors     << ORB_SLAM3:cmake /home/robot/ros_ws/logs/ORB_SLAM3/build.cmake.034.log                                                                                                                       
CMake Warning (dev) in CMakeLists.txt:
  No project() command is present.  The top-level CMakeLists.txt file must
  contain a literal, direct call to the project() command.  Add a line of
  code such as

    project(ProjectName)

  near the top of the file, but after cmake_minimum_required().

  CMake is pretending there is a "project(Project)" command on the first
  line.
This warning is for project developers.  Use -Wno-dev to suppress it.

CMake Deprecation Warning at /opt/ros/noetic/share/ros/core/rosbuild/rosbuild.cmake:7 (cmake_policy):
  The OLD behavior for policy CMP0011 will be removed from a future version
  of CMake.

  The cmake-policies(7) manual explains that the OLD behaviors of all
  policies are deprecated and that a policy should be set to OLD only under
  specific short-term circumstances.  Projects should be ported to the NEW
  behavior and not rely on setting a policy to OLD.
Call Stack (most recent call first):
  CMakeLists.txt:2 (include)

CMake Deprecation Warning at /opt/ros/noetic/share/ros/core/rosbuild/rosbuild.cmake:16 (cmake_policy):
  The OLD behavior for policy CMP0002 will be removed from a future version
  of CMake.

  The cmake-policies(7) manual explains that the OLD behaviors of all
  policies are deprecated and that a policy should be set to OLD only under
  specific short-term circumstances.  Projects should be ported to the NEW
  behavior and not rely on setting a policy to OLD.
Call Stack (most recent call first):
  CMakeLists.txt:2 (include)

CMake Deprecation Warning at /opt/ros/noetic/share/ros/core/rosbuild/rosbuild.cmake:18 (cmake_policy):
  The OLD behavior for policy CMP0003 will be removed from a future version
  of CMake.

  The cmake-policies(7) manual explains that the OLD behaviors of all
  policies are deprecated and that a policy should be set to OLD only under
  specific short-term circumstances.  Projects should be ported to the NEW
  behavior and not rely on setting a policy to OLD.
Call Stack (most recent call first):
  CMakeLists.txt:2 (include)

CMake Deprecation Warning at /opt/ros/noetic/share/ros/core/rosbuild/rosbuild.cmake:20 (cmake_policy):
  The OLD behavior for policy CMP0005 will be removed from a future version
  of CMake.

  The cmake-policies(7) manual explains that the OLD behaviors of all
  policies are deprecated and that a policy should be set to OLD only under
  specific short-term circumstances.  Projects should be ported to the NEW
  behavior and not rely on setting a policy to OLD.
Call Stack (most recent call first):
  CMakeLists.txt:2 (include)

CMake Deprecation Warning at /opt/ros/noetic/share/ros/core/rosbuild/rosbuild.cmake:23 (cmake_policy):
  The OLD behavior for policy CMP0011 will be removed from a future version
  of CMake.

  The cmake-policies(7) manual explains that the OLD behaviors of all
  policies are deprecated and that a policy should be set to OLD only under
  specific short-term circumstances.  Projects should be ported to the NEW
  behavior and not rely on setting a policy to OLD.
Call Stack (most recent call first):
  CMakeLists.txt:2 (include)

[rosbuild] Building package ORB_SLAM3
[rosbuild] Error from syntax check of ORB_SLAM3/manifest.xml
Traceback (most recent call last):
  File "/usr/lib/python3.8/encodings/__init__.py", line 31, in <module>
    import codecs
  File "/usr/lib/python3.8/codecs.py", line 96
    *, _is_text_encoding=None):
     ^
SyntaxError: invalid syntax
CMake Error at /opt/ros/noetic/share/ros/core/rosbuild/private.cmake:77 (message):
  [rosbuild] Syntax check of ORB_SLAM3/manifest.xml failed; aborting
Call Stack (most recent call first):
  /opt/ros/noetic/share/ros/core/rosbuild/public.cmake:174 (_rosbuild_check_manifest)
  CMakeLists.txt:4 (rosbuild_init)

cd /home/robot/ros_ws/build/ORB_SLAM3; catkin build --get-env ORB_SLAM3 | catkin env -si  /usr/bin/cmake /home/robot/ros_ws/src/covins/orb_slam3/Examples/ROS/ORB_SLAM3 --no-warn-unused-cli -DCATKIN_DEVEL_PREFIX=/home/robot/ros_ws/devel -DCMAKE_INSTALL_PREFIX=/home/robot/ros_ws/install -DCMAKE_BUILD_TYPE=RelWithDebInfo; cd -

..........................................................................................................................................................................................................
Failed     << ORB_SLAM3:cmake               [ Exited with code 1 ]                                                                                                                                        
Failed    <<< ORB_SLAM3                     [ 0.2 seconds ]                                                                                                                                               
[build] Summary: 7 of 8 packages succeeded.                                                                                                                                                               
[build]   Ignored:   72 packages were skipped or are blacklisted.                                                                                                                                         
[build]   Warnings:  None.                                                                                                                                                                                
[build]   Abandoned: None.                                                                                                                                                                                
[build]   Failed:    1 packages failed.                                                                                                                                                                   
[build] Runtime: 1.2 seconds total.

zhouleiqiang, 2021-12-29:
After two hours of testing, I solved this problem. The reason is that ros noetic uses python2.7 by default, so we only need to change python2.7 to python3.7. Just enter the following commands in the terminal.
`sudo apt install python-is-python3`

patriksc, 2022-01-03:
Thanks for sharing your experience with COVINS, and also the solution you found for your problem. However, I cannot fully follow your explanations - this [ROS Noetic documentation page](https://wiki.ros.org/noetic/Migration) states explicitly that _ROS Noetic will target Python 3 exclusively_

patriksc, 2022-01-08:
Issue was resolved

closed 2022-01-08 by patriksc