v4rl-ucy/covins #10: How to run covins with your own camera

zhouleiqiang, 2021-12-31:
Hello, I read the README file you wrote, but I did not find a tutorial on how to run covins with my own camera. Can you teach me?

patriksc, 2022-01-03:
Since COVINS uses ORB-SLAM3 as the VIO front-end, the instructions provided by the authors of ORB-SLAM3 might help: https://github.com/UZ-SLAMLab/ORB_SLAM3#4-running-orb-slam3-with-your-camera. However, note that they have just released a new version end of 2021, where they changed the calibration file format.

patriksc, 2022-01-08:
Could you connect your own camera in the meantime? Basically, you need to 
1. calibrate your visual-inertial sensor system, so that you can generate a calibration file similar to `orb_slam3/Examples/Monocular-Inertial/EuRoC.yaml`. You can use e.g. [kalibr](https://github.com/ethz-asl/kalibr) for calibrating. Use the _radtan_ distortion model. This [document](https://github.com/UZ-SLAMLab/ORB_SLAM3/blob/master/Calibration_Tutorial.pdf) provided by the authors of ORB-SLAM3 might help as well (note that the parameter names align perfectly after their update to v1.0)
2. Generate a launch file similar to `launch_ros_euroc.launch` that points to your calibration file and adjust the topics so that they fit your sensor system.

patriksc, 2022-07-31:
Problem seems resolved for the moment - closing this issue.

closed 2022-07-31 by patriksc