v4rl-ucy/covins #11: Segfaults reported after commit e21f25992369d39c9abb920259dd8fef3f30d973

patriksc, 2022-01-08:
Some users have reported running into a segfault after commit `e21f25992369d39c9abb920259dd8fef3f30d973` (see issue #2). The maintainers cannot reproduce this issue, therefore, this needs to be observed whether it occurs for more users in the future. Since the issue was most likely introduced by the correction of the thread handing, we a special branch [nothreadfix](https://github.com/VIS4ROB-lab/covins/tree/versions/nothreadfix) was created, which is similar to the current master, but does not have the thread handling corrections. 

When using this branch, the program will therefore still report `terminate called without an active exception` on termination, which can be ignored.

RoZhong, 2022-01-12:
Sorry I didn't experiment until today. The answer is below ，it is still error

[image]
[image]
[image]

RoZhong, 2022-01-12:
> 一些用户报告在提交后`e21f25992369d39c9abb920259dd8fef3f30d973`遇到了段错误（参见问题#2）。维护者无法重现此问题，因此需要观察未来是否会出现更多用户。由于问题很可能是由线程处理的更正引入的，因此我们创建了一个特殊的分支[nothreadfix](https://github.com/VIS4ROB-lab/covins/tree/versions/nothreadfix)，它与当前的 master 类似，但没有线程处理的更正。
> 
> 因此，当使用此分支时，程序仍会报告`terminate called without an active exception`终止，可以忽略。

but this time It runs to "wait for back-end response", I looked at the last one and it didn't run to this step

patriksc, 2022-01-18:
OK - so the error must be somewhere in the communicator. As already mentioned, since we cannot reproduce the bug, you need to perform manual debugging on your machine and figure out where and why this error is thrown. We already tried to investigate what happens in line `409` by adding some `cout` commands printing the `ContainerSize` and `msg_type_container_.size()` - I would recommend revisiting this idea. Particularly, if there is not output shown, the failure cannot be caused by line 409, since the `cout` statements are executed before. If this is the case, you need to trace down the line that actually causes the segfault.

patriksc, 2022-01-20:
The most promising way to find the problem is to deactivate the changes introduced after commit `e21f25992369d39c9abb920259dd8fef3f30d973` step by step and see which change causes the problem on your machine. 
The compare function of github might help here: https://github.com/VIS4ROB-lab/covins/compare/e21f25992369d39c9abb920259dd8fef3f30d973...versions/nothreadfix