v4rl-ucy/covins #12: experiment in real word

flyover-26, 2022-03-23:
hi！
Could you tell me whether you tested covins in real world like ccm?how is its performance?
Looking forward to your reply!

patriksc, 2022-04-14:
Yes, we have also run some real-world tests with COVINS. However, since there was no ground truth available in this setup, we do not have a quantitative performance evaluation available for those experiments.

flyover-26, 2022-04-15:
thank you!
I test this in real-world on one computer.
And sometime  I suddenly got a problem like
------- selectserver: socket 11 hung up. -------
----> Exit Recv Thread
Comm 0: close
And orb-slam front-end stops immediately.

patriksc, 2022-07-31:
Problem seems resolved for the moment - closing this issue.

closed 2022-07-31 by patriksc

South-River, 2024-06-01:
> thank you!
> I test this in real-world on one computer.
> And sometime  I suddenly got a problem like
> ------- selectserver: socket 11 hung up. -------
> ----> Exit Recv Thread
> Comm 0: close
> And orb-slam front-end stops immediately.

how's its performance in real world. Can you share me your results in real worlds.
Can it's results compete with the kimera multi or even better?
thank you:D