v4rl-ucy/covins #13: no eigen, ceres, opengv directory

JaimeParker, 2022-03-23:
Hi everyone, in file `fix_eigen_deps.sh`
```sh
#eigen_catkin
cd ${BASEDIR}/..
if [ -d "eigen_catkin" ]
then
  cd eigen_catkin
  FIX_EIGEN_VERSION="1"
  if grep -q "covins_patched" CMakeLists.txt; then
    FIX_EIGEN_VERSION="0"
  fi
#  echo "FIX_EIGEN_VERSION: $FIX_EIGEN_VERSION"
  if [ "$FIX_EIGEN_VERSION" == "1" ]
  then
    echo "set eigen_catkin to 3.3.4"
    sed -i '13d' CMakeLists.txt
    sed -i '12aset(EIGEN_MINIMUM_VERSION 3.3.4)' CMakeLists.txt
    sed -i '88d' CMakeLists.txt
    sed -i '88d' CMakeLists.txt
    sed -i '87a\ \ \ \ URL https://gitlab.com/libeigen/eigen/-/archive/3.3.4/eigen-3.3.4.tar.bz2' CMakeLists.txt
    sed -i '88a\ \ \ \ URL_MD5 6e74a04aeab3417120f1bdef6f3b4881' CMakeLists.txt
    sed -i '46aset(USE_SYSTEM_EIGEN "OFF")' CMakeLists.txt
    sed -i '1i# covins_patched' CMakeLists.txt
  else echo "eigen_catkin already set to version 3.3.4"
  fi
else echo "ERROR: no eigen_catkin directory"
fi
```
it means there should be a folder called `eigen_catkin` in `covins_ws/src`, but there isn't.
I have `eigen3.4.0` and `ceres` on my ubuntu, not sure about `opengv`(I can run ORB_SLAM3 and 2 on my device).
Do I need to copy these files in `src` or is this a bug? Or did I mistakenly operate something? thanks for your reply, in advance.

ThomasZiegler, 2022-03-23:
Hi
The mentioned folders (besides others) are listed in the `dependencies.rosinstall` file and get automatically installed from the [install file](https://github.com/VIS4ROB-lab/covins/blob/c67e968667ee32a898d4feb237c276c25bda395f/install_file.sh#L17). So if you follow the [installation guide](https://github.com/VIS4ROB-lab/covins/blob/c67e968667ee32a898d4feb237c276c25bda395f/readme.md?plain=1#L65) step by step all dependencies should be installed. 
Regards Thomas

closed 2022-03-23 by ThomasZiegler

JaimeParker, 2022-03-23:
Sir I'm rather confused, there is no command to download eigen, ceres and opengv in file `src`, so when install script pointing to `fix_eigen_deps.sh`, it just run down to `ERROR: no eigen_catkin directory`

JaimeParker, 2022-03-23:
I just git clone all files in `dependencies.rosinstall` in `src`, with the `covins` file.

JaimeParker, 2022-03-23:
fixed, need to clone all files in `dependencies.rosinstall` , then run that install command

ThomasZiegler, 2022-03-23:
It's strange that you had to clone the repos manually. Normally the installation scripts takes care of that and automatically. It clones the repos via the `python3-wstool`. Could it be that you did not install the listed [dependencies](https://github.com/VIS4ROB-lab/covins/blob/c67e968667ee32a898d4feb237c276c25bda395f/readme.md?plain=1#L76) i.e. `sudo apt-get install python3-wstool`?

JaimeParker, 2022-03-24:
@ThomasZiegler thank you, sir.  I tested `sudo apt-get install python3-wstool` again, and it said that python3-wstool has been the newest version,  and when I ran it before 
```sh
cd ${BASEDIR}/..
wstool init
wstool merge covins/dependencies.rosinstall
wstool up
```
I can see info provided by wstool, just didn't know whether it's successful.
Whatever, it did show that if something went wrong, we can install these files manually, also a solution I guess.

cccleon, 2024-04-18:
I reproduce this problem by running
`wstool merge covins/dependencies.rosinstall`
, then I get
`File "/usr/lib/python3/dist-packages/wstool/config_yaml.py", line 74, in get_yaml_from_uri
    yamldata = yaml.load(stream)
TypeError: load() missing 1 required positional argument: 'Loader'`
YAML does not support `.load(f)` after v5.1 anymore as it's considered to be unsafe. `safe_load` or `load(f, Loader=FullLoader)` is recommended.
So I just replace `yaml.load` with `yaml.safe_load` and it goes well.