v4rl-ucy/covins #15: Usage of Se(3) computations in the backend

DanielRoeder1, 2022-03-29:
I am working on adjusting COVINS to use monocular SLAM ( I am aware of the existence of CCMSLAM but it is not suitable for my work).

As mentioned in https://github.com/VIS4ROB-lab/covins/issues/2#issuecomment-973077972 these changes should mainly revolve around extending the backend from Se(3) to Sim(3) calculations.

So far I identified the place recognition routine (PlaceRecognition::ComputeSE3) as the main cooperate for these changes. 

Could someone kindly guide me towards other uses cases that I may have missed or other changes that would need to be implemented?

patriksc, 2022-04-14:
Hi Daniel
Yes that's correct, the ComputeSE3 would need to be changed to calculate a Sim(3) transform. Furthermore, you would also need to adjust the subsequent steps that perform the actual loop closure or map merge operation to operate on similarity transformations. 
Beyond that, you would probably want to adjust the GBA method from 6 DoF BA to a 7 DoF optimization.  

Finally, an important aspect to note is the following: when you merge 2 maps M_A and M_B, from 2 agents A and B, those maps will have a different scale. Place recognition will calculate am Sim(3) transformation (with scale factor **s**)  to transform M_B into M_A, i.e. the data from M_B will adopt the scale from M_A. This means that subsequently, the scale factor of the data contributed by Agent B will be different from the associated map on the server side, so that you will have to keep track of this scale factor **s** from the map merge and scale the data coming from Agent B using **s** before adding it to the merged map.

Hope that helps! Generally, I would look into the CCM-SLAM code when adjusting COVINS for monocular SLAM, there you can find those functionalities already implemented, which should provide a good starting point to port them to COVINS.

patriksc, 2022-07-31:
Problem seems resolved for the moment - closing this issue.

closed 2022-07-31 by patriksc