v4rl-ucy/covins #16: Error when running client in docker image

Tariq-Abuhashim, 2022-04-06:
Hi;    

I run the following:

In terminal 1: ./run.sh -c
In terminal 2: ./run.sh -s ../covins_comm/config/config_comm.yaml ../covins_backend/config/config_backend.yaml
In terminal 3: ./run.sh -o ../covins_comm/config/config_comm.yaml ../orb_slam3/covins_examples/euroc_examples_mh1

Then in terminal 3, I get this error:

run_copy.sh: 1: run_copy.sh: %YAML:1.0: not found
run_copy.sh: 7: run_copy.sh: sys.port:: not found
run_copy.sh: 8: run_copy.sh: sys.server_ip:: not found
run_copy.sh: 14: run_copy.sh: comm.send_updates:: not found
run_copy.sh: 15: run_copy.sh: comm.data_to_client:: not found
run_copy.sh: 17: run_copy.sh: comm.start_sending_after_kf:: not found
run_copy.sh: 18: run_copy.sh: comm.kf_buffer_withold:: not found
run_copy.sh: 19: run_copy.sh: comm.max_sent_kfs_per_iteration:: not found
run_copy.sh: 20: run_copy.sh: comm.update_window_size:: not found
run_copy.sh: 22: run_copy.sh: comm.to_agent_freq:: not found
run_copy.sh: 24: run_copy.sh: comm.start_drift_estimation_after_kf:: not found
run_copy.sh: 30: run_copy.sh: orb.activate_visualization:: not found
run_copy.sh: 31: run_copy.sh: orb.imu_stamp_max_diff:: not found

I used Docker installation, my covins filesystem looks like this

covins
├── covins_backend
│   ├── CMakeLists.txt
│   ├── config
│   ├── covins_sys
│   ├── Doxyfile.in
│   ├── include
│   ├── launch
│   ├── output
│   ├── package.xml
│   ├── src
│   ├── srv
│   └── thirdparty
├── covins_comm
│   ├── CMakeLists.txt
│   ├── config
│   ├── include
│   ├── package.xml
│   ├── src
│   └── thirdparty
├── dependencies.rosinstall
├── docker
│   ├── Dockerfile
│   ├── Makefile
│   └── run.sh
├── fix_eigen_deps.sh
├── install_file.sh
├── orb_slam3
│   ├── build_ros.sh
│   ├── build.sh
│   ├── Changelog.md
│   ├── CMakeLists.txt
│   ├── covins_examples
│   ├── Dependencies.md
│   ├── evaluation
│   ├── Examples
│   ├── include
│   ├── LICENSE
│   ├── README.md
│   ├── src
│   ├── Thirdparty
│   └── Vocabulary
├── readme.md
└── release_notes.txt

ThomasZiegler, 2022-04-20:
Hi
The `./run.sh -o` requires three parameters (`COMM_CONFIG`, `RUN_FILE`, and  `DATASET_FOLDER`). From the example you give above you are missing the `DATASET_FOLDER` (i.e. the path to the euroc data). Do you get the same error when you add the missing parameter?

patriksc, 2022-05-20:
Hello @urbste , thanks for the feedback. You are right, the path in the run.sh needs to be corrected - I'll push a fix in the next days. Thanks for pointing this out!

patriksc, 2022-05-23:
@urbste I just pushed an update for the `run.sh` file. It works fine locally on my machine - if you have a moment to verify it works on your side as well, that would be much appreciated.

patriksc, 2022-07-31:
Problem seems resolved by code update - closing this issue.

closed 2022-07-31 by patriksc