v4rl-ucy/covins #17: map reset frequently

flyover-26, 2022-04-11:
hi!
I test this with realsense d435i camera.
However,I find it will reset frequently.I konw orb-slam3 have the same problem.Is there any method to let this system run well with my own camara?
Looking forward to your reply!

patriksc, 2022-04-14:
Hi! Since this implementation of COVINS employs ORB-SLAM3 as the VIO front-end, if ORB-SLAM3 has know problems with this kind of camera, you will unfortunately also see them in COVINS.

flyover-26, 2022-04-15:
thank you so much!
I notice that you mentioned that whenmaps are not merged yet, the visualization in RVIZ might toggle between the visualization of both trajectories.
Is there any method to avoid this happening?
thank you for your patience!

patriksc, 2022-05-19:
Unfortunately, this is a minor issue with the way the visualization is implemented currently. This is indeed not optimal, we'll try fix this in the future - but for now, there's not immediate way to avoid this, sorry!

patriksc, 2022-07-31:
Problem seems resolved for the moment - closing this issue.

closed 2022-07-31 by patriksc