v4rl-ucy/covins #19: Monocular running covins error

zhouleiqiang, 2022-05-02:
**Hello, when I run your project with zed2 camera, I get an error, Failed to open settings file at:** /home/robot1/robot1_ws/covins_ws/src/covins/orb_slam3/Examples/ROS/ORB_SLAM3/zed_yaml/zed_cam2.yaml
[ORB_SLAM3_monoi0-1] process has died [pid 18301, exit code 255, cmd /home/robot1/robot1_ws/covins_ws/src/covins/orb_slam3/Examples/ROS/ORB_SLAM3/Mono /home/robot1/robot1_ws/covins_ws/src/ covins/orb_slam3/Vocabulary/ORBvoc.txt /home/robot1/robot1_ws/covins_ws/src/covins/orb_slam3/Examples/ROS/ORB_SLAM3/zed_yaml/zed_cam2.yaml /camera/image_raw:=/zed2/zed_node/left/image_rect_color __name :=ORB_SLAM3_monoi0 __log:=/home/robot1/.ros/log/21d068ce-ca1e-11ec-b096-54b203f6b90f/ORB_SLAM3_monoi0-1.log].
log file: /home/robot1/.ros/log/21d068ce-ca1e-11ec-b096-54b203f6b90f/ORB_SLAM3_monoi0-1*.log.

But I can't find the corresponding log file in the corresponding folder. The topic I subscribed to is **/zed2/zed_node/left/image_rect_color**. Do you have any solution?

patriksc, 2022-05-19:
Hello @zhouleiqiang. Since COVINS is a visual-inertial framework, it not supposed to be used with the monocular version of ORB-SLAM3. If you want to perform collaborative SLAM in the monocular domain, we would recommend you to try out our other work, CCM-SLAM: https://github.com/VIS4ROB-lab/ccm_slam
Also, for issues regarding interfacing ORB-SLAM3 with your own camera, we would recommend to also pose such questions in the ORB-SLAM3 issues: https://github.com/UZ-SLAMLab/ORB_SLAM3/issues

patriksc, 2022-07-31:
Problem seems resolved for the moment - closing this issue.

closed 2022-07-31 by patriksc