v4rl-ucy/covins #21: What happens to keyframes received by the front-end ?

Tariq-Abuhashim, 2022-05-24:
Is this source complete?
What happens to keyframes received at the front-end?
In communicator_base, looking at ``auto CommunicatorBase::ProcessBufferIn()->void `` function, nothing seems to be using ``buffer_keyframes_in`` or ``buffer_landmarks_in``.

patriksc, 2022-05-29:
Yes, this is complete. As explained in the [manual](https://github.com/VIS4ROB-lab/covins/tree/master#ext_reuse), COVINS provides the infrastructure to send data from the server back to an agent to allow the re-use information from the server onboard the agent. However, we do not provide a certain algorithm for map re-use, this needs to be implemented by the user. Unfortunately, there is currently no active work on including such an algorithm into the COVINS framework.

patriksc, 2022-07-31:
Problem seems resolved for the moment - closing this issue.

closed 2022-07-31 by patriksc