v4rl-ucy/covins #22: Corrected drift of ORB_SLAM3

WuZihao12, 2022-05-27:
Hi，I want to ask
How to use the historical key frames returned by the server to the client to correct the drift of the client (ORB_SLAM3).

patriksc, 2022-05-29:
There are several ways in the literature how this can be implemented. One simple option for example would be the following:
Assuming you have an Agent `A` with KFs `0,...,i`, where both `KF0` (which defines the origin of `A`'s coordinate frame) and `i` are present both on the agent and server side, since data is shared. Then you could communicate the (drift-corrected by BA) transform `T_0_i(s)` from the server to the agent, and compare it with the same (but drift-affected) transform that the agent sees, `T_0_i(a)`. This would give you an estimate of the drift. 

This is a very simple algorithm though - we would also encourage you to explore how different works in the literature (for example [CCM-SLAM](https://github.com/VIS4ROB-lab/ccm_slam) implement map re-use.

Also, please note that you need to take measures on the server side to correct for the drift, i.e. there should be additional information from other agents present, or a loop closure on the data from A should have been found, otherwise, the drift in the data shared by `A` cannot be corrected.

WuZihao12, 2022-05-30:
Thanks for your answer

patriksc, 2022-07-31:
Problem seems resolved for the moment - closing this issue.

closed 2022-07-31 by patriksc