v4rl-ucy/covins #25: How to run COVINS front-end on different machines connected through a wireless network ?

Janebek, 2022-07-26:
Hi ! @patriksc  I've been trying to run COVINS front-end on different machines, but it seems it does not have tutourial about this. I can see the tutourial about using my own camera to perform the fornt-end , but it seems it is all performed on one machine(fornt-end and back-end). So how to exactly do this ? I didn't any thing in CCM-SLAM tutourial either .

ThomasZiegler, 2022-07-26:
Hi Janebek

The tutorial does not state it explicitly but it works exactly the same for either running the front-end and back-end on the same device or running it on different devices. 

As written in the [README](https://github.com/VIS4ROB-lab/covins/blob/f8e6527bafffa5bba53a888b0e7ec625724baafe/readme.md?plain=1#L153) you need to set the IP of the server in the config of every agent. As long as the the IP of the server is reachable from the device the communication works.

As you can see in the paper we even tested COVINS running on an AWS server with local agents. So the communication works even over the internet and not only in a local network.

Regards Thomas

closed 2022-07-26 by ThomasZiegler

Janebek, 2022-07-27:
Hi Thomas 

Thank you so much for your reply ,  this really helps a lot .