v4rl-ucy/covins #26: Stereo-Inertia

pengllbit, 2022-08-01:
Does this support Stereo-Inertia,or how can I use covins to run my own Stereo-Inertia camera?

ThomasZiegler, 2022-08-04:
Hi
Unfortunately COVINS does not support stereo. But this doesn't mean you can't run a stereo front-end in you SLAM. But when transmitting the data from your front-end to the COVINS back-end you should only transmit the keyframes of one camera. 

To implement COVINS in your own setup you need to transmit the keyframes and map points/landmarks from your front-end to the COVINS server. For this you need to transfer the keyframes and map points into messages `covins::MsgKeyframe` and `covins::MsgLandmark`, respectively. You can use our ORB SLAM 3 implementation as a guide.
https://github.com/VIS4ROB-lab/covins/blob/f8e6527bafffa5bba53a888b0e7ec625724baafe/orb_slam3/include/KeyFrame.h#L258-L272
https://github.com/VIS4ROB-lab/covins/blob/f8e6527bafffa5bba53a888b0e7ec625724baafe/orb_slam3/include/MapPoint.h#L103-L106

Regards Thomas

pengllbit, 2022-08-05:
Thank you for your answer
I have another problem. When I use my own camera_D435i to run COVINS Mono_Inertial with ROS , it always fails to initialize. I went to see ORB_SLAM3 and failed to solve this problem. Can you give some initialization suggestions.

pengllbit, 2022-08-05:
[image]

pengllbit, 2022-08-05:
Fails to initialize like this
[image]

ThomasZiegler, 2022-08-11:
Hi, if you have another problem please open a new dedicated issue for it. Otherwise it's hard for other people, which have similar problems to find the discussion, when it's not related to the current issue title. 

However, in your case it's seems that your new problem is very similar to issue #10. As stated there we adapted ORB SLAM 3 as front-end, but we didn't change anything in the initialization part. So to get COVINS with our front-end working you essentialy need to get ORB SLAM 3 running with your camera setup. So I would follow their [instruction](https://github.com/UZ-SLAMLab/ORB_SLAM3#4-running-orb-slam3-with-your-camera) and look in their [issues](https://github.com/UZ-SLAMLab/ORB_SLAM3/issues) if some one had similar problems or otherwise open an new issue there.

patriksc, 2023-12-31:
Closing - original issue resolved

closed 2023-12-31 by patriksc