v4rl-ucy/covins #30: how to find the

XiliangLi, 2022-08-31:


patriksc, 2022-10-29:
Hi, COVINS should also find loops in the single-agents setup. However, on MH1, there's usually only a single loop closure at the very end of the sequences, connecting beginning and end of the trajectory. This loop is also importation for the global accuracy of the MH1 estimate. Does this loop also not appear when you run COVINS with a single agent on MH1?  

Generally, to trigger more loops, you can change the values for place recognition parameters in `config_backend.yaml` (`placerec.xxx`). However, pay attention that this can also lead to false positive loop closures if the thresholds are too loose.

patriksc, 2022-11-13:
If you only cut the initialization pattern (i.e. you start the MH1 before the UAV is in the air a second time, at approx. 45s), you should still see the loop on MH1.

Regarding your run on MH1, 2 and 3: there are certain conditions for loop closures, e.g. there must be a certain distance between 2 loop closures, so for example if a loop closure between agent A and B is found, there will not be an immediate subsequent one between data from A and A. As long as you find sufficient loop closures, and get an accurate global estimate, it should not matter whether they are between a single agent or different agents. Maybe the sequence in which the data is sent to the server is such that there's always a loop between 2 different agents found first - finding only loops between different agents is by itself not a defect.

patriksc, 2023-12-31:
Closing - inactive

closed 2023-12-31 by patriksc