v4rl-ucy/covins #31: failed to run in ros

selfInnovator, 2022-11-08:
hello, i have a question:
after the following steps:
In ~/ws/covins_ws/src/covins/orb_slam3/Examples/ROS/ORB_SLAM3/launch/launch_ros_euroc.launch: adjust the paths for voc and cam
cd to orb_slam3/ and run roslaunch ORB_SLAM3 launch_ros_euroc.launch
the error occurs:
ERROR: cannot launch node of type [ORB_SLAM3/Mono_Inertial]: Cannot locate node of type [Mono_Inertial] in package [ORB_SLAM3]. Make sure file exists in package path and permission is set to executable (chmod +x)
how to solve it?

ThomasZiegler, 2022-11-09:
Hi it's hard to identify the problem with the limited information you provide. Are you sure that you installed the ORB SLAM3 ROS support? It requires additional installation steps not performed in the `install_file.sh` script. For description how to install ROS support see in the readme https://github.com/VIS4ROB-lab/covins/blob/f8e6527bafffa5bba53a888b0e7ec625724baafe/readme.md?plain=1#L117-L143

selfInnovator, 2022-11-09:
thanks for your answer, i have solved the problem

closed 2022-11-09 by ThomasZiegler