v4rl-ucy/covins #33: Error build of ORB_SLAM3

aboelela924, 2023-02-24:
I ran all the commands to build ORB_SLAM3 until build vision_opencv everything ran without errors after that i had the following:

/usr/bin/ld: CMakeFiles/Mono.dir/src/ros_mono.cc.o: in function `estd2::GetStringFromYaml(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >)':
/home/aboelela/ws/covins_ws/src/covins/covins_comm/include/covins/covins_base/typedefs_base.hpp:90: undefined reference to `cv::FileStorage::FileStorage(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, int, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)'
/usr/bin/ld: /home/aboelela/ws/covins_ws/src/covins/covins_comm/include/covins/covins_base/typedefs_base.hpp:98: undefined reference to `cv::FileStorage::operator[](std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) const'
/usr/bin/ld: CMakeFiles/Mono.dir/src/ros_mono.cc.o: in function `cv::FileNode::operator std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >() const':
/usr/include/opencv4/opencv2/core/persistence.hpp:570: undefined reference to `cv::FileNode::string[abi:cxx11]() const'
/usr/bin/ld: CMakeFiles/Mono.dir/src/ros_mono.cc.o: in function `int estd2::GetValFromYaml<int>(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >)':
/home/aboelela/ws/covins_ws/src/covins/covins_comm/include/covins/covins_base/typedefs_base.hpp:76: undefined reference to `cv::FileStorage::FileStorage(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, int, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)'
/usr/bin/ld: /home/aboelela/ws/covins_ws/src/covins/covins_comm/include/covins/covins_base/typedefs_base.hpp:84: undefined reference to `cv::FileStorage::operator[](std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) const'
/usr/bin/ld: /home/aboelela/ws/covins_ws/src/covins/covins_comm/include/covins/covins_base/typedefs_base.hpp:84: undefined reference to `cv::FileNode::operator double() const'
/usr/bin/ld: CMakeFiles/Mono.dir/src/ros_mono.cc.o: in function `bool estd2::GetValFromYaml<bool>(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >)':
/home/aboelela/ws/covins_ws/src/covins/covins_comm/include/covins/covins_base/typedefs_base.hpp:76: undefined reference to `cv::FileStorage::FileStorage(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, int, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)'
/usr/bin/ld: /home/aboelela/ws/covins_ws/src/covins/covins_comm/include/covins/covins_base/typedefs_base.hpp:84: undefined reference to `cv::FileStorage::operator[](std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) const'
/usr/bin/ld: /home/aboelela/ws/covins_ws/src/covins/covins_comm/include/covins/covins_base/typedefs_base.hpp:84: undefined reference to `cv::FileNode::operator double() const'
/usr/bin/ld: CMakeFiles/Mono.dir/src/ros_mono.cc.o: in function `double estd2::GetValFromYaml<double>(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >)':
/home/aboelela/ws/covins_ws/src/covins/covins_comm/include/covins/covins_base/typedefs_base.hpp:76: undefined reference to `cv::FileStorage::FileStorage(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, int, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)'
/usr/bin/ld: /home/aboelela/ws/covins_ws/src/covins/covins_comm/include/covins/covins_base/typedefs_base.hpp:84: undefined reference to `cv::FileStorage::operator[](std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) const'
/usr/bin/ld: /home/aboelela/ws/covins_ws/src/covins/covins_comm/include/covins/covins_base/typedefs_base.hpp:84: undefined reference to `cv::FileNode::operator double() const'
collect2: error: ld returned 1 exit status
make[2]: *** [CMakeFiles/Mono.dir/build.make:214: /home/aboelela/ws/covins_ws/src/covins/orb_slam3/Examples/ROS/ORB_SLAM3/Mono] Error 1
make[1]: *** [CMakeFiles/Makefile2:741: CMakeFiles/Mono.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
/usr/bin/ld: CMakeFiles/Mono_Inertial.dir/src/ros_mono_inertial.cc.o: in function `estd2::GetStringFromYaml(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >)':
/home/aboelela/ws/covins_ws/src/covins/covins_comm/include/covins/covins_base/typedefs_base.hpp:90: undefined reference to `cv::FileStorage::FileStorage(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, int, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)'
/usr/bin/ld: /home/aboelela/ws/covins_ws/src/covins/covins_comm/include/covins/covins_base/typedefs_base.hpp:98: undefined reference to `cv::FileStorage::operator[](std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) const'
/usr/bin/ld: CMakeFiles/Mono_Inertial.dir/src/ros_mono_inertial.cc.o: in function `cv::FileNode::operator std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >() const':
/usr/include/opencv4/opencv2/core/persistence.hpp:570: undefined reference to `cv::FileNode::string[abi:cxx11]() const'
/usr/bin/ld: CMakeFiles/Mono_Inertial.dir/src/ros_mono_inertial.cc.o: in function `int estd2::GetValFromYaml<int>(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >)':
/home/aboelela/ws/covins_ws/src/covins/covins_comm/include/covins/covins_base/typedefs_base.hpp:76: undefined reference to `cv::FileStorage::FileStorage(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, int, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)'
/usr/bin/ld: /home/aboelela/ws/covins_ws/src/covins/covins_comm/include/covins/covins_base/typedefs_base.hpp:84: undefined reference to `cv::FileStorage::operator[](std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) const'
/usr/bin/ld: /home/aboelela/ws/covins_ws/src/covins/covins_comm/include/covins/covins_base/typedefs_base.hpp:84: undefined reference to `cv::FileNode::operator double() const'
/usr/bin/ld: CMakeFiles/Mono_Inertial.dir/src/ros_mono_inertial.cc.o: in function `bool estd2::GetValFromYaml<bool>(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >)':
/home/aboelela/ws/covins_ws/src/covins/covins_comm/include/covins/covins_base/typedefs_base.hpp:76: undefined reference to `cv::FileStorage::FileStorage(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, int, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)'
/usr/bin/ld: /home/aboelela/ws/covins_ws/src/covins/covins_comm/include/covins/covins_base/typedefs_base.hpp:84: undefined reference to `cv::FileStorage::operator[](std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) const'
/usr/bin/ld: /home/aboelela/ws/covins_ws/src/covins/covins_comm/include/covins/covins_base/typedefs_base.hpp:84: undefined reference to `cv::FileNode::operator double() const'
/usr/bin/ld: CMakeFiles/Mono_Inertial.dir/src/ros_mono_inertial.cc.o: in function `double estd2::GetValFromYaml<double>(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >)':
/home/aboelela/ws/covins_ws/src/covins/covins_comm/include/covins/covins_base/typedefs_base.hpp:76: undefined reference to `cv::FileStorage::FileStorage(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, int, std::__cxx1 [... cut ...]  const'
/usr/bin/ld: /home/aboelela/ws/covins_ws/src/covins/covins_comm/include/covins/covins_base/typedefs_base.hpp:84: undefined reference to `cv::FileNode::operator double() const'
collect2: error: ld returned 1 exit status
make[2]: *** [CMakeFiles/Mono_Inertial.dir/build.make:214: /home/aboelela/ws/covins_ws/src/covins/orb_slam3/Examples/ROS/ORB_SLAM3/Mono_Inertial] Error 1
make[1]: *** [CMakeFiles/Makefile2:172: CMakeFiles/Mono_Inertial.dir/all] Error 2
make: *** [Makefile:130: all] Error 2

patriksc, 2023-03-04:
Hi! Could you share some more details n your system (e.g. Ubuntu version, special hardware used)? Also, I see from your logs that you are using OpenCV4 - did you follow the steps listed [here](https://github.com/VIS4ROB-lab/covins#installing-ros-support-for-the-orb-slam3-front-end) under _If you are running Ubuntu 20 (or generally have OpenCV 4 installed)_ ?

aboelela924, 2023-03-04:
Hi, I followed the steps yes. I was using a virutal machine using ubuntu 20.04 on vmware.

patriksc, 2023-03-17:
Hey, I have now tested this on my side (VirtualBox VM w/ latest Ubuntu 20.04 image). It worked smoothly for me, I did not see such an error. I copied all commands directly from the readme. I have seen this error earlier though, and it usually appeared when the OpenCV version was not aligned, i.e. linking against OpenCV 4. Given that t worked alright for me, my best guess here is that you might have missed a step from the readme, and I would suggest  to try the deployment again with a clean workspace.

patriksc, 2023-12-31:
Closing - inactive

closed 2023-12-31 by patriksc

MF-Ahmed, 2024-05-01:
Hi I am facing the same problem on Ubunto 20.04, OvenCV 4, ROS Noetic. Did you manage to find the solution ?