v4rl-ucy/covins #35: Unable to make covins_ Backend succeeded

cchester25, 2023-03-13:
There are many "Undefined reference", when i make covins_backend , i dont know how to take it .
[image]

patriksc, 2023-03-17:
Hi, what environment are you using? Which command are you executing when this error occurs? I have just tested the deployment of COVINS on a 20.04 machine, I could deploy the code without running into error.

chocho2222, 2023-04-07:
@patriksc First of all thank you for your work. I also encountered the same problem, I am using ubuntu18.04, when I run "./src/covins/install_file.sh 8", this error will appear, I will attach my screenshot below
[image]

[image]
[image]
[image]

chocho2222, 2023-04-07:
> There are many "Undefined reference", when i make covins_backend , i dont know how to take it . [image]
Hello bro,Is this problem solved?

chocho2222, 2023-04-09:
@cchester25 
Bro, I solved this problem by modifying glog_catkin versio
![Screenshot from 2023-04-09 16-59-53](
[image]
https://user-images.githubusercontent.com/91018212/230764259-b5a8257b-4bd7-4950-a3f4-1b49910e3bf7.png)
n

patriksc, 2023-04-10:
@chocho2222 Thanks a lot for sharing you experience and solution of the problem. Could you give some more details - which version of `glog` were you using, and what changes did you have to make to `glog_catkin`? I'll also try on a fresh 18.04 machine the next days, to see whether I can reproduce the error there.

patriksc, 2023-04-16:
Thanks for the additional information. I'll test in on a virtual machine with 18.04 then as well.

patriksc, 2023-12-31:
Closing. Works on VM with 18.04.

closed 2023-12-31 by patriksc