v4rl-ucy/covins #38: How to use the catkin project to build COVINS_G+vinsfusion，just like orbslam3+covins_G?

NatureLan-sudo, 2023-04-24:
As the title suggests, the code currently adapted for the VINSFusion front-end is provided in the form of Docker. May I ask how to build covins_G+vinsfusion directly on the local machine, just like covins_G+ORBSLAM3 as described in the readme? Thank you very much for your help.

manthan99, 2023-04-24:
Hi

I would suggest to use the commands provided in this docker file for reference:
https://github.com/manthan99/VINS-COVINS-adaptation/blob/master/docker/Dockerfile
 
I will try to put the instructions here:

Install the following dependencies:
```
sudo apt-get update && apt-get install -y \
      cmake \
      libatlas-base-dev \
      libeigen3-dev \
      libgoogle-glog-dev \
      libsuitesparse-dev \
      python-catkin-tools \
      ros-melodic-cv-bridge \
      ros-melodic-image-transport \
      ros-melodic-message-filters \
      ros-melodic-tf
```

1. Create the workspace: `mkdir -p ~/catkin_ws/src`
2. `cd ~/catkin_ws && catkin init`
3. `catkin config --extend /opt/ros/melodic/`
4. `catkin config --cmake-args -DCMAKE_BUILD_TYPE=RelWithDebInfo`
5. Clone the repository: `cd ~/catkin_ws/src && git clone https://github.com/manthan99/VINS-COVINS-adaptation.git`
6. Clone the following repositories as well :
 ```git clone https://github.com/ethz-asl/ceres_catkin.git && \
    git clone https://github.com/catkin/catkin_simple.git && \
    git clone https://github.com/ethz-asl/opencv3_catkin.git && \
    git clone https://github.com/ethz-asl/eigen_catkin.git && \
    git clone https://github.com/ethz-asl/glog_catkin.git
```
7.  `cd ~/catkin_ws && catkin build`

Let me know if the build is successful and then I can let you know the further steps

Best
Manthan

NatureLan-sudo, 2023-04-24:
Hi, @manthan99 , Thank you very much for the help. I compared the dockfile and found that one library was missing from the steps you provided. It is as follows:
`git clone https://github.com/ethz-asl/ceres_catkin.git && \
    git clone https://github.com/catkin/catkin_simple.git && \
    git clone https://github.com/ethz-asl/opencv3_catkin.git && \
    git clone https://github.com/ethz-asl/eigen_catkin.git && \
    git clone https://github.com/ethz-asl/glog_catkin.git
    `
In addition, I used the **fix_eigen_deps.sh** script provided in the COVINS  to set the library version. Otherwise, there might be a problem with not being able to find the file in my environment. Currently, I have successfully built the project.
 May I know what else do I need to do after this? Thank you very much for your help. I will also provide feedback to you actively.

Best 
Nature

NatureLan-sudo, 2023-04-24:
Hi, @manthan99 , Thank you very much for the help. I compared the dockfile and found that one library was missing from the steps you provided. It is as follows:
```
git clone https://github.com/ethz-asl/ceres_catkin.git 
git clone https://github.com/catkin/catkin_simple.git 
git clone https://github.com/ethz-asl/opencv3_catkin.git 
git clone https://github.com/ethz-asl/eigen_catkin.git 
git clone https://github.com/ethz-asl/glog_catkin.git 
```
In addition, I used the **fix_eigen_deps.sh** script provided in the COVINS to set the library version. Otherwise, there might be a problem with not being able to find the file in my environment. 
Currently, I have successfully built the project.
May I know what else do I need to do after this? Thank you very much for your help. I will also provide feedback to you actively.

Best
Nature

manthan99, 2023-04-24:
Ah yes, you definitely need ceres_catkin. I somehow copied it incorrectly. I am glad that you were able to build the repository. 

Now, you need to simply use the relevant launch file. For example if you want to run on Euroc Dataset, you will launch the following (Make sure your workspace is sourced, communication config is correct (https://github.com/manthan99/VINS-COVINS-adaptation/blob/master/covins_comm/config/config_comm.yaml#L14) and the backend is already running):

Ag0: `roslaunch vins vins_covins.launch config:=${PATH TO CONFIG file} ag_n:=0`
Ag1: `roslaunch vins vins_covins.launch config:=${PATH TO CONFIG file} ag_n:=0`

You can check this for reference:
https://github.com/manthan99/VINS-COVINS-adaptation/blob/master/docker/run_vins_covins_euroc.sh#L48

Let me know if this works

NatureLan-sudo, 2023-04-24:
Hi~@manthan99 . Currently, I met this problem: I have already compiled the ORBSLAM3+convins_G project, so I successfully enabled the backend in this workspace and got the correct output. Later, I enabled the frontend in the just-built Vins project, and I have modified the IP address as you suggested. The backend is currently running normally, and both sub-end frontends have started running normally. But the communication between them is failing, so the backend is not receiving any information from the frontend. I saw the following output in the terminal:
```
--> Connect to server
----> Server IP: 127.0.0.1
----> Port     : 9033
client: connect: Connection refused
client: failed to connect
FATAL: Could no establish connection - exit
```
The server IP is my local machine IP, which I have successfully used in the ORBSLAM3+COVINS_G project before.
 Is there anything I missed during the execution?

manthan99, 2023-04-26:
Hi

Good to hear that you were able to solve the problem. I would be happy to get feedback on the other modes and front-end interfaces.

You may close the issue if you feel that the problem is solved.

Best
Manthan

closed 2023-04-26 by NatureLan-sudo

zwq-slamer, 2024-10-16:
Hi!  What datasets you've run?  It's ok when i am running EuRoc , but it crashed when i ran other datasets.