v4rl-ucy/covins #39: unable to detect loops

swj6030829, 2023-04-27:
I would like to ask, I am using my own camera and the various proxies and backend are running smoothly. The local maps of each proxy can be generated successfully in the backend. However, it is still unable to detect loops regardless of how high the similarity of the scene is. May I know why or which parameters should I modify?

swj6030829, 2023-04-27:
Hi, just to add, the framework I am using is COVINS-G.

swj6030829, 2023-04-27:
[image]
[image]
Hello, this is my running result. I ran the first frontend and the second frontend separately on two ROSBAG datasets recorded in the same room with high scene similarity. I used the RGBD mode of ORBSLAM3 as my frontend, but I don't think this affects the data transmission to the backend. I published image frames and odometry information to the backend through ROS, and the backend can receive them, but it cannot detect any loop closures. Is it because there are too few keyframes generated, or some other reasons? I look forward to your reply.

manthan99, 2023-04-27:
Hi

Are you using the Front-end ROS wrapper with RGBD ORBSLAM3? From the Rviz trajectories, they do not look good and seem to be very discontinuous (should be much smoother). 

Could you enable the Keyframe visualization in rviz and resend the screenshots ?

swj6030829, 2023-04-27:
[image]
Hello, I am using the default settings and parameters. It seems that only from the third proxy node onwards, the keyframes can be seen in RVIZ. I have checked the dataset I recorded and found that the RGB publishing frequency is only 10 frames. I can adjust it to a higher frequency. Will this affect the detection of loop closures?

manthan99, 2023-04-27:
Hi

It seems like you are indeed using the Front-end ROS Wrapper that we provide.

Unfortunately, this will not work with ORB-SLAM3 directly because of the local optimizations that ORB-SLAM3 does, the frame is always drifting (until initialized properly). If you wish to use ORB SLAM3 in RGBD mode, I suggest to do modifications to the ORB-SLAM3 codebase itself (Like it is done for the monocular case). 

You can have a look at this for reference: https://github.com/manthan99/covins_g#interfacing-a-custom-vio-system-with-covins

patriksc, 2023-12-31:
Closing - inactive

closed 2023-12-31 by patriksc