v4rl-ucy/covins #40: issue in building the covins_backend of Covins-g

ajxdhe, 2023-04-28:


ajxdhe, 2023-04-28:
Excuse me
I met the issue when I build the Covins_g
..........................................................................................................................................................................................................
Failed << covins_backend:make [ Exited with code 2 ]
Failed <<< covins_backend [ 54.9 seconds ]
[build] Summary: 17 of 18 packages succeeded.
[build] Ignored: 17 packages were skipped or are skiplisted.
[build] Warnings: 2 packages succeeded with warnings.
[build] Abandoned: None.
[build] Failed: 1 packages failed.
[build] Runtime: 1 minute and 6.0 seconds total.
So I ran cd /home/skywalker/ws/covins_ws/build/covins_backend; catkin build --get-env covins_backend | catkin env -si  /usr/bin/make --jobserver-auth=3,4; cd - 
Then I met the issue that

[ 0%] Built target std_srvs_generate_messages_eus
[ 0%] Built target tf2_msgs_generate_messages_nodejs
[ 0%] Built target tf2_msgs_generate_messages_lisp
[ 0%] Built target actionlib_msgs_generate_messages_py
[ 0%] Built target actionlib_msgs_generate_messages_cpp
[ 0%] Built target actionlib_generate_messages_py
[ 0%] Built target actionlib_generate_messages_nodejs
[ 0%] Built target actionlib_generate_messages_cpp
[ 0%] Built target _covins_backend_generate_messages_check_deps_ServiceLoadMap
[ 0%] Built target _covins_backend_generate_messages_check_deps_ServicePruneMap
[ 0%] Built target _covins_backend_generate_messages_check_deps_ServiceGBA
[ 0%] Built target _covins_backend_generate_messages_check_deps_ServiceSaveMap
[ 0%] Built target roscpp_generate_messages_nodejs
[ 0%] Built target tf_generate_messages_nodejs
[ 0%] Built target tf_generate_messages_lisp
[ 0%] Built target tf_generate_messages_cpp
[ 0%] Built target roscpp_generate_messages_lisp
[ 0%] Built target topic_tools_generate_messages_nodejs
[ 0%] Built target topic_tools_generate_messages_lisp
[ 0%] Built target roscpp_generate_messages_cpp
[ 0%] Built target topic_tools_generate_messages_eus
[ 0%] Built target topic_tools_generate_messages_cpp
[ 0%] Built target actionlib_msgs_generate_messages_eus
[ 0%] Built target dynamic_reconfigure_generate_messages_py
[ 0%] Built target std_msgs_generate_messages_lisp
[ 0%] Built target sensor_msgs_generate_messages_cpp
[ 0%] Built target tf2_msgs_generate_messages_cpp
[ 0%] Built target dynamic_reconfigure_generate_messages_lisp
[ 0%] Built target std_srvs_generate_messages_lisp
[ 0%] Built target dynamic_reconfigure_gencfg
[ 0%] Built target pcl_msgs_generate_messages_cpp
[ 0%] Built target nodelet_generate_messages_cpp
[ 0%] Built target pcl_ros_gencfg
[ 0%] Built target std_srvs_generate_messages_py
[ 0%] Built target rosgraph_msgs_generate_messages_eus
[ 0%] Built target std_msgs_generate_messages_py
[ 0%] Built target geometry_msgs_generate_messages_eus
[ 0%] Built target std_msgs_generate_messages_cpp
[ 0%] Built target rosgraph_msgs_generate_messages_nodejs
[ 0%] Built target tf_generate_messages_eus
[ 0%] Built target rosgraph_msgs_generate_messages_lisp
[ 0%] Built target std_msgs_generate_messages_nodejs
[ 0%] Built target nodelet_generate_messages_lisp
[ 0%] Built target actionlib_generate_messages_lisp
[ 0%] Built target roscpp_generate_messages_py
[ 0%] Built target std_srvs_generate_messages_cpp
[ 0%] Built target rosgraph_msgs_generate_messages_py
[ 0%] Built target topic_tools_generate_messages_py
[ 0%] Built target roscpp_generate_messages_eus
[ 0%] Built target dynamic_reconfigure_generate_messages_cpp
[ 0%] Built target actionlib_msgs_generate_messages_nodejs
[ 0%] Built target actionlib_msgs_generate_messages_lisp
[ 0%] Built target tf_generate_messages_py
[ 0%] Built target std_msgs_generate_messages_eus
[ 0%] Built target actionlib_generate_messages_eus
[ 0%] Built target rosgraph_msgs_generate_messages_cpp
[ 0%] Built target geometry_msgs_generate_messages_cpp
[ 0%] Built target geometry_msgs_generate_messages_lisp
[ 0%] Built target dynamic_reconfigure_generate_messages_eus
[ 0%] Built target geometry_msgs_generate_messages_nodejs
[ 0%] Built target geometry_msgs_generate_messages_py
[ 0%] Built target tf2_msgs_generate_messages_eus
[ 0%] Built target sensor_msgs_generate_messages_py
[ 9%] Built target covins_backend_generate_messages_eus
[ 9%] Built target nodelet_generate_messages_eus
[ 9%] Built target nodelet_generate_messages_nodejs
[ 9%] Built target bond_generate_messages_nodejs
[ 9%] Built target std_srvs_generate_messages_nodejs
[ 9%] Built target nodelet_generate_messages_py
[ 9%] Built target tf2_msgs_generate_messages_py
[ 9%] Built target dynamic_reconfigure_generate_messages_nodejs
[ 9%] Built target bond_generate_messages_cpp
[ 9%] Built target bond_generate_messages_eus
[ 9%] Built target sensor_msgs_generate_messages_nodejs
[ 9%] Built target bond_generate_messages_lisp
[ 9%] Built target nodelet_topic_tools_gencfg
[ 9%] Built target sensor_msgs_generate_messages_eus
[ 9%] Built target sensor_msgs_generate_messages_lisp
[ 9%] Built target bond_generate_messages_py
[ 9%] Built target pcl_msgs_generate_messages_eus
[ 9%] Built target pcl_msgs_generate_messages_nodejs
[ 9%] Built target pcl_msgs_generate_messages_lisp
[ 9%] Built target pcl_msgs_generate_messages_py
[ 17%] Built target covins_backend_generate_messages_nodejs
[ 25%] Built target covins_backend_generate_messages_lisp
[ 32%] Built target covins_backend_generate_messages_cpp
[ 42%] Built target covins_backend_generate_messages_py
[ 53%] Built target covins_basey
[ 53%] Built target covins_backend_generate_messages
make[2]: *** No rule to make target '/home/skywalker/ws/covins_ws/src/covins_g/covins_backend/thirdparty/DBoW2/lib/libDBoW2.so', needed by '/home/skywalker/ws/covins_ws/devel/lib/libcovins_backend.so'. Stop.
make[1]: *** [CMakeFiles/Makefile2:324: CMakeFiles/covins_backend.dir/all] Error 2
make: *** [Makefile:141: all] Error 2

My system is ubuntu 20.04

ajxdhe, 2023-04-28:
Actually I followed the instuction you mentioned,but the issue occured in the final step --building the covins_backend 
I builded the covins  sucessfully before ,but I failed to build the covins_g .

manthan99, 2023-04-28:
Hi

Probably the issue was caused because I had renamed the repository without taking into account some of the paths in the install script. I have now switched back to the original name for now and thus it should not be an issue anymore. Thus I suggest reinstalling everything in a fresh workspace. Let me know if you still encounter any errors.

closed 2023-07-14 by ThomasZiegler