v4rl-ucy/covins #41: Failed to run the vins fronted in COVINS-G

ajxdhe, 2023-04-28:
I tried to modified the `vins_covins.launch` to use my realsense camera.
This is the detail of launch file 
<launch>
<arg name="ag_n" default="0" />
<arg name="config" default="/home/skywalker/ws2/vins/src/VINS-COVINS-adaptation/config/d455/vins_fusion_realsense_d455.yaml" />

<group ns="agent_$(arg ag_n)">
    <node pkg="loop_fusion" type="loop_fusion_node" name="loop_fusion_node" args="$(arg config)" output="screen">
        <remap from="/vins_estimator/extrinsic" to="/agent_$(arg ag_n)/vins_node/extrinsic"/>
        <remap from="/vins_estimator/keyframe_imu" to="/agent_$(arg ag_n)/vins_node/keyframe_imu"/>
        <remap from="/vins_estimator/keyframe_point" to="/agent_$(arg ag_n)/vins_node/keyframe_point"/>
        <remap from="/vins_estimator/keyframe_pose" to="/agent_$(arg ag_n)/vins_node/keyframe_pose"/>
        <remap from="/vins_estimator/margin_cloud" to="/agent_$(arg ag_n)/vins_node/margin_cloud"/>
        <remap from="/vins_estimator/odometry" to="/agent_$(arg ag_n)/vins_node/odometry"/>
        <remap from="/camera/infra1/image_rect_raw" to="/cam0/image_raw$(arg ag_n)"/>
        <remap from="/camera/infra2/image_rect_raw" to="/cam1/image_raw$(arg ag_n)"/>
        <remap from="/camera/imu" to="/imu$(arg ag_n)"/>
    </node>

    <node pkg="vins" type="vins_node" name="vins_node" args="$(arg config)" output="screen">
        <remap from="/camera/infra1/image_rect_raw" to="/cam0/image_raw$(arg ag_n)"/>
        <remap from="/camera/infra2/image_rect_raw" to="/cam1/image_raw$(arg ag_n)"/>
        <remap from="/camera/imu" to="/imu$(arg ag_n)"/>
        <remap from="/agent_$(arg ag_n)/vins_node/camera_pose" to="/agent_$(arg ag_n)/cam_odom"/>
    </node>
</group>

</launch>

but I got the error 
[agent_0/vins_node-2] process has died [pid 5509, exit code -11, cmd /home/skywalker/ws2/vins/devel/lib/vins/vins_node /home/skywalker/ws2/vins/src/VINS-COVINS-adaptation/config/d455/vins_fusion_realsense_d455.yaml /camera/infra1/image_rect_raw:=/cam0/image_raw0 /camera/infra2/image_rect_raw:=/cam1/image_raw0 /camera/imu:=/imu0 /agent_0/vins_node/camera_pose:=/agent_0/cam_odom __name:=vins_node __log:=/home/skywalker/.ros/log/d28a065e-e5f5-11ed-b4ed-03ba6cf212c6/agent_0-vins_node-2.log].
log file: /home/skywalker/.ros/log/d28a065e-e5f5-11ed-b4ed-03ba6cf212c6/agent_0-vins_node-2*.log

manthan99, 2023-05-02:
Hi

Could you please send the full launch file that you are using? As a reference, you can probably use this launch file which was made for realsense camera (You will need to change vins_covins.launch according to this). 
https://github.com/manthan99/VINS-COVINS-adaptation/blob/master/vins_estimator/launch/vins_frontend_realsense.launch

Moreover, are you able to run on EuRoC datasets ?

Best
Manthan

manthan99, 2023-10-27:
Hi

I am closing this issue due to inactivity. Feel free to reopen it if it didn't solve your problem

closed 2023-10-27 by manthan99