v4rl-ucy/covins #42: Is it ok to use my own RGBD dataset?

TwiceMao, 2023-05-11:
Thank you for such a great job！
Is it ok to use my own RGBD dataset? I have a mutli-room RGBD dataset, but I don't have IMU, LiDAR, Stereo and other data, can your code run on this dataset?
Besides, the RGB image of my RGBD dataset is in .jpg format, the Depth image is in .png format, and the pose is in .txt format. Will this still work with your work? If not, how can I convert the data format to meet the requirements?
Thank you for your reply！

manthan99, 2023-05-15:
Hi

Yes, ideally it should be possible to be able to run it with [COVINS-G](https://github.com/manthan99/covins). In order to do this, you will need to have a good RGB-D front-end. For example, if you are using ORB-SLAM3 RGBD front-end, you will need to modify its codebase so as to send the KFs in the correct way to the back-end via the communication module. You can have a look at these instructions for reference (Current changes to ORB-SLAM3 are for mono-inertial and you will have to do similar changes for the RGB-D setting).  

https://github.com/VIS4ROB-lab/covins#ext_comm

Alternatively, if you some other RGBD Front-end where the front-end estimates are not optimized everytime, you can also try directly using the COVINS-G with the ROS wrapper which will require minimal/no changes to the codebase. 

https://github.com/manthan99/covins/blob/master/docs/run_COVINS-G.md#run_fe_ros

Note that running ORB-SLAM3 with the ROS wrapper will not work since ORB_SLAM3 performs local optimizations which cause the trajectory to move. To use ORB-SLAM3, you will need to modify the codebase as stated above.

Best
Manthan

patriksc, 2023-12-31:
Closing - inactive

closed 2023-12-31 by patriksc