v4rl-ucy/covins #43: Hello, I have deployed VINS-FUSION+ego_planner on my drone equipment. In this case, can I connect to the covsin-g backend to collaborate?

yjjzz12619, 2023-05-14:


manthan99, 2023-05-15:
Hi @yjjzz12619 

Yes, you should be able to use VINS-fusion with the COVINS-G back-end. You can find the running instructions here:

https://github.com/manthan99/covins/blob/master/docs/run_COVINS-G.md#run_fe_vins

Though I would recommend to go through the full documentation for correctly installing everything.

Best
Manthan

yjjzz12619, 2023-08-01:
Thank you, but what I mean is that I am unable to install the VINS-COVINS-adaptation you provided. If I want to use Covins-g on the Vins-Fusion I have already installed, what modifications do I need to make to the files?

manthan99, 2023-08-07:
Hi @yjjzz12619 

Could you please let me know what issues are you having while installing the VINS-COVINS-adaptation, since it would be much easier to use that rather than modifying the VINS codebase?

An alternative might be to use the original VINS and the front-end wrapper with that (Though you won't have the map points in this case). You can find the instructions here: 
https://github.com/VIS4ROB-lab/covins/blob/master/docs/run_COVINS-G.md#run_fe_ros
You can have a look at the instructions for SVO-Pro and do similar stuff for your VINS version.

Best
Manthan

yjjzz12619, 2023-09-01:
>

yjjzz12619, 2023-09-01:
Hi@manthan99
 This is an error I encountered when installing VINS-COVINS-adaptation using Docker.
[image]

yjjzz12619, 2023-09-01:
If I use the original VINS and the front end wrapper, which parts of Covins should I install on the drone end? And how can I run it so that my drone doesn't use the dataset but its own camera? Thank you very much for your answer!

yjjzz12619, 2023-09-01:
[image]

yjjzz12619, 2023-09-02:
Hello, I tried VINS+d435i and the front end wrapper, but unfortunately it still failed. The backend seems to be unable to receive vins data, and some of my configurations are as follows.
[image]

manthan99, 2023-09-06:
> Hi@manthan99 This is an error I encountered when installing VINS-COVINS-adaptation using Docker. [image]

Regarding this, I would recommend opening a separate issue on the VINS-COVINS-adaptation repo and provide more information/logs so that I can have a look

manthan99, 2023-09-06:
> If I use the original VINS and the front end wrapper, which parts of Covins should I install on the drone end? And how can I run it so that my drone doesn't use the dataset but its own camera? Thank you very much for your answer!

For this, you need to follow the instructions for running with a front-end ROS wrapper which are provided here:
https://github.com/VIS4ROB-lab/covins/blob/master/docs/run_COVINS-G.md#run_fe_ros

You can use the instructions for SVO-pro to understand what modifications might be needed to your launch files of VINS (Because the instructions/scripts that we provide for VINS are for our version of modified VINS with docker only)

While building covins on your drone, you can only run the following command to exclusively build only the ROS-based Frontend-wrapper:
`./src/covins/install_file.sh -f 8`

To Summarize you will need to do the following on your drone:

1. Build the covins repo with front-end wrapper only
2. Install a working version of VINS-fusion and modify the launch/config file for VINS (`rosrun vins vins_node ~/catkin_ws/src/VINS-Fusion/config/euroc/euroc_mono_imu_config.yaml`). Specifically, you will need to modify the input topics of the camera/IMU according to your setup/dataset.
3. Modify the topics in COVINS-front-end ROS wrapper launch file (https://github.com/VIS4ROB-lab/covins/blob/master/covins_frontend/launch/vins_euroc_agent.launch). The cam topic is the input topic for the camera, the odom topic is the odometry output topic provided by your VINS node. Lastly you might need to modify the config file according to the camera params, IMU-Camera TF, etc.

So you will need to run the launch files of Step 2,3 on your Drone. Make sure that the config file for communication has the correct IP set (https://github.com/VIS4ROB-lab/covins/blob/master/covins_comm/config/config_comm.yaml) on the drone.

I hope this makes it clear. If not feel free to put additional queries. I would recommend to run the full setup on the EuRoC dataset before you try your custom setup.

manthan99, 2023-09-06:
> Hello, I tried VINS+d435i and the front end wrapper, but unfortunately it still failed. The backend seems to be unable to receive vins data, and some of my configurations are as follows. [image]

Could you check the RQT plots to make sure that all the topic names are correct? 

Also what is the terminal output of the front-end wrapper launch file ?

Did you set the correct IP address between the agent and the server ?

patriksc, 2023-12-31:
Closing - inactive

closed 2023-12-31 by patriksc