v4rl-ucy/covins #45: Map Preservation and Map Reuse

LanZY2021, 2023-07-14:
Hello, thank you very much for your open source project. CoVINS is a framework for multi-robot mapping to build sparse visual feature point maps. But the framework does not seem to be able to save the map, nor can the saved map be used for robot relocation. May I ask, will this part of the functionality be updated subsequently?

ThomasZiegler, 2023-07-14:
Hi

As described in the Readme in section [User Interaction](https://github.com/VIS4ROB-lab/covins/blob/master/docs/run_COVINS.md#run_intercation) the maps can be saved per agent via rosservice call. 
One can then load the maps at a later time. If then the current active agent visits a part of the loaded map a map merge should be triggered. However, keep in mind that the map merging happens only on the server side, i.e. the information is not sent back to the agent.

LanZY2021, 2023-07-14:
Thank you very much for your reply. I would like to discuss with you about map reuse. Sparse visual point maps are often not able to describe the three-dimensional structural environment well, and are more commonly used for "visual localization based on prior maps". Will CoVINS support the function of "visual localization based on prior maps" in the future? For example, Maplab supports offline map fusion and "visual localization based on prior maps

patriksc, 2023-07-16:
>  Sparse visual point maps are often not able to describe the three-dimensional structural environment well
I wouldn't agree with this statement, this depends heavily on the use case. For large-scale SLAM and visual localization, sparse graph-based maps are among the best performing approaches in the literature. 

> Will CoVINS support the function of "visual localization based on prior maps" in the future?
COVINS is mainly designed and intended as a collaborative SLAM system, that allows multiple agents to collaborate online during the mission. Adding functionalities such as visual localization and offline map fusion would definitely be great extension to the system, but there are currently no active plans to release such functionalities.

patriksc, 2023-12-31:
Closing - inactive

closed 2023-12-31 by patriksc