v4rl-ucy/covins #49: backend segmentfault

ggbond964, 2023-09-14:
When I use a t265 camera to run the ORB_SLAM3 frontend,the covins beackend reports the segmentfault without any reasons.What might cause the segmentfault when using the t265 camera? 
The following are the frontend and backend logs when segmentfault happens.
frontend:
[image]
backend:
[image]

manthan99, 2023-10-27:
Hi

Could you please provide more details. Are you using the COVINS backend or COVINS-G backend ? 

Is your ORB-SLAM3 front-end initializing properly or failing ? The seg fault occurs in the back-end when the Front-end Map is not initialized properly or a Map-reset occurs

patriksc, 2023-12-31:
Closing - inactive

closed 2023-12-31 by patriksc