v4rl-ucy/covins #50: Use my camera to transmit data

hyc-yuchen, 2023-09-20:
How do I use my front-end camera /dev/video0 to pass data to the front-end and then view the composition information in the visualization window？

manthan99, 2023-09-25:
Hi

What exactly do you mean by the composition information. I think that will depend on the front-end that you use.

Best
Manthan

hyc-yuchen, 2023-10-09:
I use a webcam that transmits images through an rstp stream

> Hi
> 
> What exactly do you mean by the composition information. I think that will depend on the front-end that you use.
> 
> Best Manthan

manthan99, 2023-10-09:
Hi

To answer shortly, you will need to publish the images and the IMU data on rostopics. This might be something that could be useful for the images: http://wiki.ros.org/video_stream_opencv. 

Best
Manthan

hyc-yuchen, 2023-10-10:
> Hi
> 
> To answer shortly, you will need to publish the images and the IMU data on rostopics. This might be something that could be useful for the images: http://wiki.ros.org/video_stream_opencv.
> 
> Note: The backend expects the inputs to be scaled metrically and hence you would either need an mono-IMU or a stereo setup for this. Not sure if you can just use a webcam for this.
> 
> If you are interested in monocular only, have a look at CCM SLAM repository (https://github.com/VIS4ROB-lab/ccm_slam)
> 
> Best Manthan

thank you for your replay

I imported the rstp stream into the virtual camera /dev/video0 and published it on rostopics using usb_cam. Now I constructed the related Mono file by modifying the camera in ros_mono.cc and used it to connect and communicate with the covins back-end. Although the front end can see the camera collecting data, but the back end does not receive relevant information, and rviz does not have relevant information. This means that it does not transmit data to the backend. It was constructed locally.

How can I fix it

hyc-yuchen, 2023-10-23:
> Hi
>
> To answer shortly, you will need to publish the images and the IMU data on rostopics. This might be something that could be useful for the images: http://wiki.ros.org/video_stream_opencv.
>
> Note: [...]

hello ，could teach me how to do it？

manthan99, 2023-10-27:
Please check the CCM SLAM Repository  (https://github.com/VIS4ROB-lab/ccm_slam), where the detailed instructions are provided.

closed 2023-10-27 by manthan99