v4rl-ucy/covins #51: Invalid Image Dims: "- 1|-1”

chocho2222, 2023-09-26:
I replaced ORB_SLAM3-V0.4 in Covins-G with V1.0, and then ran the RGB-D dataset (rgbd_dataset_freiburg3_walking_halfsphere). Every time I ran to 160-200 pictures, the error "Invalid Image Dims: "- 1|-1””, please tell me how to solve this error. Thank you very much!

          The following two pictures show errors when running to 177 and 194 pictures respectively.
[image]
[image]

chocho2222, 2023-09-27:
[image]

chocho2222, 2023-09-27:
[image]
And there is an error: "Segmentation fault (core dumped)"

manthan99, 2023-10-27:
Hi

Just to clarify, the issue that you encounter is in the ORB-SLAM front-end, right  (Which cause the backend to crash)? 

Could you verify if the same sequence works correctly on the non-modified ORB-SLAM3 V1.0 repository ?

chocho2222, 2023-11-08:
Sorry, I saw the news very late. I didn't deal with this problem for a month. Then I ran it again and it could run through the data set. But I recompiled ORB_SLAM3 and this problem appeared again. Next, I will follow your instructions. Comments to test run

chocho2222, 2023-11-08:
> Hi
> 
> Just to clarify, the issue that you encounter is in the ORB-SLAM front-end, right (Which cause the backend to crash)?
> 
> Could you verify if the same sequence works correctly on the non-modified ORB-SLAM3 V1.0 repository ?
Sorry, I saw the news very late. I didn't deal with this problem for a month. Then I ran it again and it could run through the data set. But I recompiled ORB_SLAM3 and this problem appeared again. Next, I will follow your instructions. Comments to test run

patriksc, 2023-12-31:
Closing - inactive

closed 2023-12-31 by patriksc