v4rl-ucy/covins #52: Restart the backend resume

hyc-yuchen, 2023-10-16:
If I stop the backend halfway through, how do I continue to use the previous map data

csv KF_1_ftum.csv KF_2_ftum.csv KF_3_ftum.csv in covins_ws/src/covins/covins_backend/output/ The four files stamped_traj_estimate.txt are mounted. Now, how to read and compose them after restarting the backend?

manthan99, 2023-10-27:
Hi

You will need to save the map when you pause. You can then load the saved map data for your new run. The detailed instructions for this are provided here-
https://github.com/VIS4ROB-lab/covins/blob/master/docs/run_COVINS.md#run_intercation

Check the Map Save and Load feature

hyc-yuchen, 2023-10-28:
> Hi
> 
> You will need to save the map when you pause. You can then load the saved map data for your new run. The detailed instructions for this are provided here- https://github.com/VIS4ROB-lab/covins/blob/master/docs/run_COVINS.md#run_intercation
> 
> Check the Map Save and Load feature

I see the instruction "rosservice call /covins_savemap <AGENT_ID>",
What is the agent id? I see that after merging the maps associated with agent 0 and agent 1, the combined map will be associated with both agents at the same time, does this mean that if I have 5 agents transferring at the same time, do I just set AGENT_ID to any value between 0 and 4 while saving
And if I import the map data to.... /covins_backend/output/map_data folder. Do I only need to use rosservice call /covins_loadmap 0 to read.

manthan99, 2023-10-30:
Yes that seems right.

hyc-yuchen, 2023-10-31:
> Yes that seems right.

Sorry，i still have some questions，

I used "rosservice call /covins_savemap 0" at the back end to output mapdata data, but after I restart the container and run "rosservice call /covins_loadmap 0", although "retval: 0 ", but the point cloud map is not available through rviz

manthan99, 2023-11-08:
If I remember correctly, if you are using COVINS-G back-end there might not be any map-points to load since the back-end works without making use of any map points. If you are using COVINS, they should be available. 
Could you provide more information like the terminal out of the back-end ?

Best
Manthan

hyc-yuchen, 2023-11-13:
> If I remember correctly, if you are using COVINS-G back-end there might not be any map-points to load since the back-end works without making use of any map points. If you are using COVINS, they [...]

I used covins. After the data set was transferred to the front end, I saved the data. After restarting the back end of covins, I could not view the map data using rviz, and I could only see the saved data and the newly transferred data after the new front end began to transfer data

patriksc, 2023-12-31:
Hi @hyc-yuchen , I just tried this again on my end, and everything works as expected. With a my local version of COVINS, on 20.04, I did the following:

- Adjust environment paths, as described [here](https://github.com/VIS4ROB-lab/covins/blob/master/docs/run_COVINS.md#setting-up-the-environment)
- Run `roscore` (separate terminal)
- Execute `tf. launch` and start RVIZ, as described [here](https://github.com/VIS4ROB-lab/covins/blob/master/docs/run_COVINS.md#visualization) (2 separate terminals)
- Start the COVINS server (separate terminal)
- Run `euroc_examples_mh1.sh` (separate terminal). Trajectory and map points are visible in RVIZ
- After `euroc_examples_mh1.sh` is finished, run `rosservice call /covins_savemap 0` --> map is saved to` /home/pschmuck/ws/covins_ws/src/covins/covins_backend/output/map_data`. There is now a file `KF_0_ftum.csv`, and a folder `map_data` which holder the actual map data
- I stop the backend (`ctrl+c`) and reset RVIZ
- I start the backend again
- I load the map with `rosservice call /covins_loadmap 0` --> I can again see the trajectory and map points in RVIZ
Are you doing anything different than that? Could you try to replicate these steps, and see whether the visualization then works for you.