v4rl-ucy/covins #53: Docker build failure

fabiannagel, 2023-10-16:
Hey - I'm trying to build COVINS-G on Docker. I ran into some issues that I was able to fix, the build fails due to a `C++ internal compiler error`.

Here's what I did:
- The Dockerfile's base image `melodic-desktop-bionic` doesn't exist anymore on Docker Hub, so I replaced it with `melodic-desktop-full-bionic`.
- I added `--platform=linux/amd64` since I'm running Apple Silicon.
- Run `make build NR_JOBS=14`, which causes `C++ internal compiler error`.
- Reduce to `NR_JOBS=4`, only to run into the same issue.

My Docker VM resources should suffice (a limit of 12 CPU & 20 GiB memory).

The Dockerfile issue suggests that the build hasn't been looked at for a while. Is there something I can do to help it get back on track?

manthan99, 2023-10-27:
Hi

Is the error related to OpenCV ? Also could you provide more details for the error message. While it has not been tested, ideally it should be possible to run with ROS-Noetic as well. Maybe you can try with that and see if it works.

patriksc, 2023-12-31:
Closing - inactive

closed 2023-12-31 by patriksc