v4rl-ucy/covins #57: Question about back-end data transmission on a custom dataset

imdsafi09, 2023-11-22:
Hello, thank you for your work.

I have been running COVINS(G) on ROS melodic on Ubuntu 18.04 with no problem with the VINS-FUSION (ROS versions) when run on [EuRoC dataset](https://projects.asl.ethz.ch/datasets/doku.php?id=kmavvisualinertialdatasets). However, when I try to run it with our bag filles (real sense-d435i). The Vins-fusion seems working but it is unable to communicate with the ROS-wrapper (as shown in the picture).
I followed the following instructions:
1-Compiled  [VINS-COVINS-adaptation](https://github.com/manthan99/VINS-COVINS-adaptation) 
2-  setup the server IP address in ~/catkin_ws/src/VINS-COVINS-adaptation/covins_comm/config/config_comm.yaml
3-roslaunch vins vins_frontend_realsense.launch (change topic names accordingly)
4- roslaunch covins_frontend vins_d455_agent.launch(change topic names accordingly)

 Could you please let me what point I am missing?
Thank you
[image]

Ws1234567879789, 2023-11-24:
May I ask if you did not report an error when compiling VINS-COVINS-adaptation? Some of the download addresses he provided seem to be unable to download. If you compile successfully, I hope you can provide me with an answer. Thank you

manthan99, 2023-11-24:
Hi @imdsafi09 

Could you please provide more information on all the terminal outputs. For example in the terminal where you are running the back-end, are you able to establish the connections with the agents ?

There are basically 2 ways you can run the setup with your custom camera. 
- Option 1: Use the VINS-Fusion node without the front-end wrapper which directly communicates with the back-end. This was the method that you used for running the EuRoC dataset. I would recommend using this approach. In order to use this, you will need to do modifications to the following launch file : https://github.com/manthan99/VINS-COVINS-adaptation/blob/master/vins_estimator/launch/vins_covins.launch (If you are using docker, make sure that the config file is correctly selected in the https://github.com/manthan99/VINS-COVINS-adaptation/blob/master/docker/run_vins_covins_euroc.sh file). In the launch file you will need to set the camera and IMU topics according to your realsense camera as well as change the config file to your realsense camera config file.

- Option 2: This is the version which makes use of the ROS Front-end wrapper. I think this is the method that you have been trying at the moment. The instructions are provided here: https://github.com/VIS4ROB-lab/covins/blob/master/docs/run_COVINS-G.md#run_d455. I will need more information from you in order to help you debug. For example, the terminal outputs from all the terminals

Best
Manthan

imdsafi09, 2023-11-30:
Hi @manthan99 

Thank you for your detailed response. I fixed the issue, and everything was fine it was just a topic mismatch in the "covins_frontend/launch/vins_d455_agent.launch" file. 

I have another related question when I use RVIZ  while using the EuRoC dataset I can visualize the point cloud but when I am running it with my sensor input there is either no point or sometimes only a few. It is something related to my sensor data (d435i) or do I need to tweak any configuration parameters.

imdsafi09, 2023-11-30:
hi @Ws1234567879789 

First, I tried to build via docker, but it failed (several errors). Then I opened the docker file https://github.com/manthan99/VINS-COVINS-adaptation/blob/master/docker/Dockerfile, ran all the commands, and compiled it manually.

Ws1234567879789, 2023-11-30:
run all the commands, and compiled it manually .I don't quite understand how to operate this, and I'm not very familiar with Docker. Can you explain in detail how to operate it? Thank you very much

---- Replied Message ----
| From | Muhammad ***@***.***> |
| Date | 11/30/2023 13:22 |
| To | VIS4ROB-lab/covins ***@***.***> |
| Cc | Ws1234567879789 ***@***.***>,
Mention ***@***.***> |
| Subject | Re: [VIS4ROB-lab/covins] Question about back-end data transmission on a custom dataset (Issue #57) |

hi @Ws1234567879789

First, I tried to build via docker, but it failed (several errors). Then I opened the docker file https://github.com/manthan99/VINS-COVINS-adaptation/blob/master/docker/Dockerfile, ran all the commands, and compiled it manually.