v4rl-ucy/covins #58: Data interaction issues

WuZihao12, 2023-11-22:
Can a single client pass in multiple images to the covins-g backend? For example, apply our multi-camera system front end and use your covins-g back end.

manthan99, 2023-11-24:
Hi

Unfortunately the COVINS/COVINS-G back-ends are designed to receive and perform computations only on a single image provided by the front-end. However, you can still use your multi-camera system front-end which will give you better estimates, and then just send one of the images to the back-end and that should be supported. 

Let me know if you need any more specific information for this.

Best
Manthan

clean-shero, 2023-12-19:
Hello, my front-end is also a binocular image. How can I modify it? Please let me know specifically which aspects of the communication module and backend module need to be modified

clean-shero, 2023-12-19:
> Hi
>
> Unfortunately the COVINS/COVINS-G back-ends are designed to receive and perform computations only on a single image provided by the front-end. However, you can still use your multi-camera system [...]

Hello, my front-end is also a binocular image. How can I modify it? Please let me know specifically which aspects of the communication module and backend module need to be modified