v4rl-ucy/covins #63: RUN Mono Faild!

zzk525, 2023-12-19:
What is the reason why the SLAM3 front-end crashes when I run a monocular bag package and it initializes successfully? This is the error prompt word Docker/covins_ Ws # roslaunch ORB_ SLAM3 launch_ ROS_ Mono. launch
Loading ORB Vocabulary This could take a while
Vocabulary loaded!
Creation of new map with id: 0
Creation of new map with last KF id: 0
Seq Name:
Camera Parameters:
-Camera: Pinhole
-Fx: 616.526
-Fy: 614.274
-Cx: 320.662
-Cy: 238.145
-K1: 0.144774
-K2: -0.330136
-P1: 0.00143315
-P2: -0.00285275
-K3: 0
-Fps: 30
-Color order: RGB (ignored if grayscale)
ORB Extractor Parameters:
-Number of Features: 1500
-Scale Levels: 8
-Scale Factor: 1.2
-Initial Fast Threshold: 9
-Minimum Fast Threshold: 5
>>>COVID-19: Initialize communicator
++++++++++System++++++++++
Server_ IP: 172.17.0.5
Port: 9033
++++++++++Communication++++++++++
Send_ Updates: 1
Data_ To_ Client: 0
Start_ Sending_ After_ Kf: 1
Kf_ Buffer_ Within: 5
Max_ Send_ Kfs_ Per_ Iteration: 2
Update_ Window_ Size: 5
To_ Agent_ Freq: 5
++++++++++ORB-SLAM3++++++++++
Activate_ Visualization: 1
IMU_ Stamp_ Max_ Diff: 2
-->Connect to server
Newfd_: ten
-->Connected
>>>COVINS: Start comm thread
>>>COVID-19: wait for back end response
-->Set Client ID: 4
>>>COVINS: client id: 4
Gtk Message: 19:02:47.163: Failed to load module "canberra gtk module"
First KF: 0; Map init KF: 0
New Map created with 329 points
Point distribution in KeyFrame: left ->329-- right ->0
Failed to track local map!
Failed to track local map!
SYSTEM ->Resting active map in monthly case
LM: Active map reset received
LM: Active map reset, waiting
LM: Resting current map in Local Mapping
COVINS: reset kf_ Out_ Buffer_
LM: End resetting Local Mapping
LM: Reset free the mutex
LM: Active map reset, Done!!!
MnFirstFrameId=75
MnInitialFrameId=0
2 Frames set to loss
First KF: 0; Map init KF: 0
New Map created with 162 points
Point distribution in KeyFrame: left ->162-- right ->0
Failed to track local map!
Failed to track local map!
SYSTEM ->Resting active map in monthly case
LM: Active map reset received
LM: Active map reset, waiting
LM: Resting current map in Local Mapping
COVINS: reset kf_ Out_ Buffer_
LM: End resetting Local Mapping
LM: Reset free the mutex
LM: Active map reset, Done!!!
MnFirstFrameId=114
MnInitialFrameId=77
4 Frames set to loss
First KF: 0; Map init KF: 0
New Map created with 262 points
Point distribution in KeyFrame: left ->262-- right ->0
Failed to track local map!
Failed to track local map!
SYSTEM ->Resting active map in monthly case
LM: Active map reset received
LM: Active map reset, waiting
LM: Resting current map in Local Mapping
COVINS: reset kf_ Out_ Buffer_
LM: End resetting Local Mapping
LM: Reset free the mutex
LM: Active map reset, Done!!!
MnFirstFrameId=118
MnInitialFrameId=116
6 Frames set to loss
First KF: 0; Map init KF: 0
New Map created with 269 points
Point distribution in KeyFrame: left ->269-- right ->0
Process has died [PID 16907, exit code -11, cmd/home/zzk/Docker/covins_ws/src/covins/orb_slam3/Examples/ROS/ORB3LAM3/Mono/home/zzk/Docker/covins_ws/src/covins_comm/../orb_slam3/Vocabulary/ORBvoc.txt/home/zzk/Docker/covins_ws/src/covins/covins_comm/../orb_slam3/Examples/Monocular/Mono_ugv_zzk.yaml/camera/imageraw:=/camera/color/imageraw __ name:=ORB-SLAM3_mono1 __ log:=/root/. ros/log/d7742164-9e72-11ee -863b-0242ac110005/ORB_ SLAM3_ Mono1-1. log]
Log file:/root/. ros/log/d7742164-9e72-11ee-863b-0242ac110005/ORB_ SLAM3_ Mono1-1 *. log
All processes on machine have died, roslauch will exit
Shutting down processing monitor
... shutting down processing monitor complete
Done
This is the launch file:
<? XML version="1.0"?>
<launch>
<arg name="agn" default="1"/>
<arg name="voc" default="$(find covins_comm)/../orb_slam3/Vocabulary/ORBvoc. txt"/>
<arg name="cam" default="$(find covins_comm)/../orb_slam3/Examples/Monocular/Mono_ugv_zzk. yaml"/>
<node pkg="ORB-SLAM3" type="Mono" name="ORB-SLAM3_monoi $(arg agn)" args="$(arg voc) $(arg cam)" output="screen">
<remap from="/camera/imageraw" to="/camera/color/imageraw"/>
</node>
</launch>
This is the configuration file:% YAML: 1.0
#--------------------------------------------------------------------------------------------
#Camera Parameters Adjust them!
#--------------------------------------------------------------------------------------------
Camera. type: "PinHole"
#Camera calibration and destruction parameters (OpenCV)
Camera. fx: 616.5256737097483
Camera. fy: 614.2737833318839
Camera.cx: 320.6623096191914
Camera. cy: 238.14000131287267
Camera. k1: 0.1447743999036075
Camera. k2: -0.33013564518115346
Camera.p1: 0.00143314614103631
Camera.p2: -0.0028527506337560913
Camera. k3: 0.0
Camera. width: 640
Camera. height: 480
#Camera frames per second
Camera. fps: 30.0
#Color order of the images (0: BGR, 1: RGB. It is ignored if images are grayscale)
Camera RGB: 1
#--------------------------------------------------------------------------------------------
#ORB Parameters
#--------------------------------------------------------------------------------------------
#ORB Extract

manthan99, 2023-12-24:
Hi

Could you provide details of the bag files/dataset that you are using. It seems like ORB SLAM frontend is not able to handle it. 

Best
Manthan

zzk525, 2024-01-04:
> Hi
> 
> Could you provide details of the bag files/dataset that you are using. It seems like ORB SLAM frontend is not able to handle it.
> 
> Best Manthan
zzk@zzk:~/Dataset/my$ rosbag info walk_full_water.bag 
path:        walk_full_water.bag
version:     2.0
duration:    4:35s (275s)
start:       Dec 08 2023 14:19:28.02 (1702016368.02)
end:         Dec 08 2023 14:24:03.41 (1702016643.41)
size:        7.1 GB
messages:    77464
compression: none [8258/8258 chunks]
types:       gnss_comm/GnssEphemMsg         [b4346f99e431c870e392a438684c0fe8]
             gnss_comm/GnssGloEphemMsg      [6893dee0d8b1c9325a6693f3488e0144]
             gnss_comm/GnssMeasMsg          [0eafeb530a3e6637722c4e296734c119]
             gnss_comm/GnssPVTSolnMsg       [d18171357d7a159f76d4d7c0b12fb631]
             gnss_comm/GnssTimePulseInfoMsg [de306035e295e3c4b1e0771a216b495a]
             sensor_msgs/Image              [060021388200f6f0f447d0fcd9c64743]
             sensor_msgs/Imu                [6a62c6daae103f4ff57a132d6f95cec2]
             sensor_msgs/NavSatFix          [2d3a8cd499b9b4a0249fb98fd05cfa48]
topics:      /camera/color/image_raw          8257 msgs    : sensor_msgs/Image             
             /camera/imu                     55150 msgs    : sensor_msgs/Imu               
             /ublox_driver/ephem                 3 msgs    : gnss_comm/GnssEphemMsg        
             /ublox_driver/glo_ephem            12 msgs    : gnss_comm/GnssGloEphemMsg     
             /ublox_driver/range_meas         4589 msgs    : gnss_comm/GnssMeasMsg         
             /ublox_driver/receiver_lla       4589 msgs    : sensor_msgs/NavSatFix         
             /ublox_driver/receiver_pvt       4589 msgs    : gnss_comm/GnssPVTSolnMsg      
             /ublox_driver/time_pulse_info     275 msgs    : gnss_comm/GnssTimePulseInfoMsg

zzk525, 2024-01-04:
> Hi
> 
> Could you provide details of the bag files/dataset that you are using. It seems like ORB SLAM frontend is not able to handle it.
> 
> Best Manthan

The dataset was recorded using a d435i, I use monocular topics, using orbslam3 alone it is possible to run through without encountering problems, when using covins it crashes at the moment of initialization

patriksc, 2024-04-28:
I don't think we have tested COVINS with the monocular version of ORB-SLAM3, since we are targeting a visual-inertial and potentially stereo sensor setups with COVINS.