v4rl-ucy/covins #64: Where Are The Keyframe Images

htchr, 2023-12-24:
Hello,

I'm trying to use the images from keyframes as a way to sub-sample the input image feeds.
I have the euroc example working; however it doesn't seem like the keyframe or the MsgKeyframe objects actually saves the images.
I added this code to the covins_backend/keyframe_be.cpp file in the main constructor:

```cpp
Keyframe::Keyframe(MsgKeyframe msg, MapPtr map, VocabularyPtr voc) 
...
    img_ = msg.img;
    // added code to check if img exists
    std::cout << std::to_string(msg.img.rows) << std::endl;
    // save image
    // cv::imwrite("path", msg.img);
```

But it always shows that the images have a row size of 0.
I can't seem to find any setting that would change this, or anywhere where the image might be deleted.
I'm confused why these classes would have this field if it's never used.

Does anyone know how I could retrieve the images from the keyframes?

Thank you,

manthan99, 2023-12-24:
Hi

We don't send the images from the frontend to the backend but rather just the keypoints and descriptors to save on communication.
 
You can have a look into the frontend code and assign the img in the keyframe msg in order to obtain the desired behavior.

Let me know if you are not able to find it and I can look into the details. 

Best
Manthan

htchr, 2023-12-25:
Hey Manthan, 

Thanks for the quick reply!
The communication bandwidth consideration makes sense.
For anyone else curious, I ended up saving the images inside of covins/orb_slam3/KeyFrame.cc:

```cpp
auto KeyFrame::ConvertToMsg(covins::MsgKeyframe &msg, KeyFrame *kf_ref, bool is_update, size_t client_id)->void {
    ...
    if (!is_update) {
    
        cv::Mat img = imgLeft;
        // added code to save image
        std::string path = "/path/kf_" + std::to_string(mTimeStamp) + ".jpg";
        bool savdImg = cv::imwrite(path, img);
```

closed 2023-12-25 by htchr