v4rl-ucy/covins #68: Fail to run stereo using VINS-COVINS-adaptation

WCaiL, 2024-03-11:
I used euroc_stereo_config.yaml to run ./run_vins_frontend_euroc.sh and was unable to run the Euroc Stereo dataset successfully. The terminal displayed that process has died. If I want to use stereo, where should I modify the VINS-COVINS-adaptation code?

TakShimoda, 2024-05-10:
I have the exact same issue. Running their VINS fork for docker with stereo. Tried it on euroc and my own dataset and they both crash with a message like this with no helpful information:

```
[agent_0/vins_node-2] process has died [pid 632, exit code -11, cmd /root/catkin_vins/devel/lib/vins/vins_node /root/catkin_vins/src/VINS-Fusion/config/euroc/euroc_stereo_config.yaml /cam0/image_raw:=/cam0/image_raw0 /cam1/image_raw:=/cam1/image_raw0 /imu0:=/imu0 /agent_0/vins_node/camera_pose:=/agent_0/cam_odom __name:=vins_node __log:=/root/.ros/log/9949e35e-0ee5-11ef-9112-9da1b0f3228b/agent_0-vins_node-2.log].
log file: /root/.ros/log/9949e35e-0ee5-11ef-9112-9da1b0f3228b/agent_0-vins_node-2*.log
```

zwq-slamer, 2024-10-16:
I have the same issue. Have you figured out?

WCaiL, 2024-11-11:
> I have the same issue. Have you figured out?

I think it's a problem with the communication module. I found that both the sender and receiver require IMU data. 
Using covins_frontend and COVINS-G backend, it will work well in stereo mode.